Xinyuan-LilyGO/T5S3-4.7-e-paper-PRO #3: SD card errors

ripplebiz, 2024-11-28:
I cannot get the SD card access working in my own code. The SD.begin() fails most times, and even the few times it returns success, every SD.open() fails.

Also, I noticed with 'display_demo.ino' sketch, in the 'Test' screen, the SD icon is shown on the 'fail' side (right side) sometimes. I found reproduceable steps are to power on device, put SD card in, then press the RESET button. You'll then see the SD as fail.
If you eject the SD card, push it back in, then press RESET, the SD will be OK. 

Also, I already tried setting the LORA_CS and SD_CS to HIGH at the start of setup(). This didn't seem to make any difference.

Here is some logging output when trying the SD.open():

[ 45997][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 46097][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 46197][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 46297][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 46297][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[ 46299][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 46403][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 46503][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 46603][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 46603][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[ 46604][E][vfs_api.cpp:105] open(): /sd/tiles/9/257/256.png does not exist, no permits for creation

ShallowGreen123, 2024-11-28:
can you show me your code, i am using here unrecoverable for your question

ripplebiz, 2024-11-28:
I just got the SD to work, by moving the digitalWrite(CD_CS, HIGH) (and the LORA_CS) right at the start of setup().
I had this running slightly later in the setup() sequence, but before the SD.begin(), and it didn't work. The timing seems to be very sensitive.
Also, I still get the problem of SD fail if you reset without ejecting then re-inserting the SD card. This is reproducable every time.

The sx1262.begin() also reports success, but am not receiving any packets. I've double checked the LoRa settings, like SF, and just never receives. Also tried sending packet, and it also reports success, but receiving device doesn't hear anything.

ripplebiz, 2024-11-28:
OK, I found the problem. If I comment out the two lines below (from setup()) the the LoRa radio works, but if they're left in then no packets:

  pinMode(BL_EN, OUTPUT);   // BL_EN is resolving to (40)
  analogWrite(BL_EN, 5);

closed 2025-01-12 by github-actions[bot]