Xinyuan-LilyGO/T5S3-4.7-e-paper-PRO #7: What is the difference between the 2 new trees? Are there 2 different hardware versions?

DualTachyon, 2025-01-24:
Hi,

I noticed that the master branch was split into 2 trees and it seemingly appears there are 2 different hardware releases of the T5 S3 EP Pro:

H752-01 | v1.0-241224 | v1.2_250118 | latest
H752 | v1.0-240810 | v1.0-241203 | current

How does one know which version of the hardware one has?
Thanks.

ShallowGreen123, 2025-01-25:
Yes, we will introduce a new version H752-01, compared to the previous version, with some improvements：
- Adds the ink screen power management chip TPS65185;
- Supports local refresh, can adjust the Vcom voltage control display color depth;
- Support [epdiy](https://github.com/vroland/epdiy) v7 direct drive.
- Added GPS module;

You can tell which version is on or off the GPS; The H752-01 has GPS, the H752 doesn't;

H752-01 is not available yet, H752-01 has a better display;

DualTachyon, 2025-01-25:
Thanks for the clarification.

closed 2025-01-25 by DualTachyon

MMT-TLVassociation, 2025-02-25:
I need help for PT4K
Mr. DualTachyon, can you help me

My email address is optics@dj9.so-net.ne.jp