Xinyuan-LilyGO/T5S3-4.7-e-paper-PRO #10: GPS Not Working

stanhamil, 2025-03-21:
I have the latest T5 S3 Pro with GPS, but no satellites are ever picked up. I’ve had it out in the open for hours, and it finds nothing.

I assume my unit is faulty and need to return it. How do I do this?

thadius83, 2025-03-29:
Same issue.

b8b8, 2025-03-29:
Mine was working, then i tried to install the M5 Launcher. Then reverted back to stock and the GPS stopped working.

[ 4502][I][SensorCommon.tpp:65] begin(): Using Arduino Wire interface.
[ 4502][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[ 5468][I][SensorCommon.tpp:65] begin(): Using Arduino Wire interface.
[ 5468][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[ 5469][I][TouchDrvGT911.hpp:583] initImpl(): GT911 using 0xBA address!
[ 5516][I][TouchDrvGT911.hpp:601] initImpl(): Product id:911
[ 5516][I][TouchDrvGT911.hpp:607] initImpl(): Firmware version: 0x1060
[ 5520][I][TouchDrvGT911.hpp:609] initImpl(): Resolution : X = 540 Y = 192
[ 5523][I][TouchDrvGT911.hpp:610] initImpl(): Vendor id:57
[ 5529][I][TouchDrvGT911.hpp:611] initImpl(): Refresh Rate:20 ms
[ 5534][I][TouchDrvGT911.hpp:612] initImpl(): MaxTouchPoint:2
[ 5541][I][TouchDrvGT911.hpp:631] initImpl(): Interrupt Mode: LOW_LEVEL_QUERY
Init GT911 Sensor success!
[ 5565][D][TouchDrvGT911.hpp:464] reloadConfig(): reloadConfig check_sum : 0xD0
[SX1262] Initializing ... success!
E (6669) gpio: gpio_install_isr_service(450): GPIO isr service already installed
All settings succesfully changed!
[SX1262] Sending first packet ... [SX1262] recv task Suspend ... SD Card Type: SDHC

Try to init L76K . Wait stop .
L76K GNSS init succeeded, using L76K GNSS Module
GPS Task Create...!

LVGL Init
LVGL UI Entry
h=10, m=19, s=6
[ 10190][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[input] X:0 Y:0
[input] X:281 Y:579
code=7

[7] gps is clicked.

[input] X:281 Y:579
No GPS detected: check wiring.
No GPS detected: check wiring.
No GPS detected: check wiring.

b8b8, 2025-03-29:
> I have the latest T5 S3 Pro with GPS, but no satellites are ever picked up. I’ve had it out in the open for hours, and it finds nothing.
> 
> I assume my unit is faulty and need to return it. How do I do this?

I think you just have a firmware issue.

stanhamil, 2025-03-30:
> > I have the latest T5 S3 Pro with GPS, but no satellites are ever picked up. I’ve had it out in the open for hours, and it finds nothing.
> > I assume my unit is faulty and need to return it. How do I do this?
> 
> I think you just have a firmware issue.

I’ve reinstalled the stock firmware from this GitHub, so if the issue is with the firmware, **LilyGo** have an issue with it, not me.

b8b8, 2025-03-30:
> I’ve reinstalled the stock firmware from this GitHub, so if the issue is with the firmware, **LilyGo** have an issue with it, not me.

Yup. Out of the box the GPS worked fine and picked up 7 satellites at my window.
I tried to install the M5 launcher but it didnt work. 
I installed the Firmware posted here (from 2 months ago) and the everything but the GPS worked again.
I assume its just a typo or bug in the firmware here vs the older ones they shipped with.

stanhamil, 2025-03-30:
> Yup. Out of the box the GPS worked fine and picked up 7 satellites at my window. I tried to install the M5 launcher but it didnt work. I installed the Firmware posted here (from 2 months ago) and [...]

I guess that would make sense.

b8b8, 2025-04-07:
New firmware was posted 5 days ago
time to give it a shot.

b8b8, 2025-04-08:
I can confirm the GPS works again with the 1.6 firmware. 

[image]

[image]

stanhamil, 2025-04-11:
> I can confirm the GPS works again with the 1.6 firmware.
> 
> [image]
> 
> [image]

Where did you find that version? I can only see v1.2…

b8b8, 2025-04-11:
where did you look?
its right in the firmware folder.
https://github.com/Xinyuan-LilyGO/T5S3-4.7-e-paper-PRO/tree/H752-01/firmware

stanhamil, 2025-04-11:
I found it just before you replied! I wonder why it isn’t the version listed on the main page?

I’ve flashed the new version and my device is now picking up GPS; the time is being updated correctly despite the number of satellites showing as 0 👍🏻

bmorcelli, 2025-04-22:
> I tried to install the M5 launcher but it didnt work.

H752-01 is not compatible with the actual Launcher.. I need to make some Library adaptations to make Launcher work with this new hardware 

Once I have something, I'll let you know

closed 2025-06-06 by github-actions[bot]